RoboticsClubAtUCDavis/micromouse #18: Create classes for hardware components

blhough, 2017-03-02:
The [hardware](https://docs.google.com/spreadsheets/d/1v4yZ6iHlCe61Mf9AZj_Za-24lgkbLBGUx5Sx-UTaCA4/edit#gid=0) for the mouse has been ordered so we need to start planning that portion of the code.

For now let's just create an empty class for each component that we will be interfacing with.

Create empty class:
- [ ] Motor
- [ ] Encoder
- [ ] RangeFinder

Additionally we will want a class to encapsulate all the components.
I am not sure a good name for this class (Hardware??)
- [ ] <Good class name here>

daniel9a, 2017-03-02:
@blhough  In what file should the empty classes go?

blhough, 2017-03-02:
Each will have their own header and cpp file

blhough, 2017-03-02:
View comment on dcf2b617b9a1cb13876d5bec564ea51ec4820848

blhough, 2017-03-02:
Only the class definition goes in the header. The function definitions go in the cpp

blhough, 2017-03-02:
The class definition for Motor is still in the cpp

closed 2017-03-04 by blhough