RoboticsClubAtUCDavis/micromouse #21: add a function for getting the angle between directions

blhough, 2017-03-02:
May want an absolute version and a relative version

ex. angleDiff( N, W) = 90
angleDiff( W, N) = -90
angleDiffAbs( W, N) = 90

may want it to return an enum since the values are discrete
may want to overload operator
may want to use better names

ids1024, 2017-03-16:
@cnagda Did you try to push this to `dev`? If so it didn't work; the repository is now configured so you can't push directly to `dev`, but changes have to go through a pull request.

klanmiko, 2017-03-16:
For the person implementing this:

Measure the angle clockwise from argument 1 to argument 2

blhough, 2017-03-16:
@cnagda  checkout the [readme](https://github.com/RoboticsClubAtUCDavis/micromouse/blob/dev/README.md) for upto instructions on cintributing

closed 2017-05-01 by blhough