RoboticsClubAtUCDavis/micromouse #43: Implement Mapping Strategy 2

blhough, 2017-03-09:
Try to come up with a maze mapping algorithm that accomplishes the following:
- Finds the Finish (*Required*)
- Minimize the time spent mapping (approximately equal to the number of moves)
- Minimize the path from start to finish
- Not grossly inefficient

Feel free to add new helper functions as needed.

Make a function called `void mapMazeStrategy2()` in the Mouse class to put you implementation in.

klanmiko, 2017-03-09:
How would this be different from mapping strategy 3?

blhough, 2017-03-09:
It's unlikely two people will come up with the same algorithms.

klanmiko, 2017-03-13:
@blhough is calling move() sufficient to guarantee that the robot will be at the location, or should I add error checking

blhough, 2017-03-13:
``` C++   
void moveTo(CellCoordinate position);
void followPath(const Path &path);

// If `keepGoing` is true the mouse will not stop at the end of the
// movement. If `useCaution` is true it will stop when a wall is
// encountered. Returns the total number of nodes traveled.
unsigned move(DirectionVector movement, bool keepGoing, bool useCaution);
```
So given our initial interface (which you can feel free to change/expand as needed)

if you call `move()` with `useCaution=false` it should be because you are certain the mouse can move there.
if you call it with `useCaution=true` you may find it convenient to use the return value, but you can assume the mouse coordinates (and direction) are always correct. eg. if you tell the mouse to move 3 but it stops at a wall after 2, the coordinates will have only changed by 2.

`moveTo` and `followPath` right now assume that the path is guaranteed clear, but it may be useful to add a `useCaution` flag to those as well.

blhough, 2017-03-13:
Additionally you might find it useful to add a isVisited flag to the nodes/cells to keep track of where you've been