RoboticsClubAtUCDavis/micromouse #47: Use Neural Network and Genetic Algorithm to derive mapping algorithm

blhough, 2017-03-09:
I don't want us to get detracted us from the main project but it would be awesome if we had this.

Here's an example.
https://www.youtube.com/watch?v=qv6UVOQ0F44

It uses this algorithm
http://nn.cs.utexas.edu/downloads/papers/stanley.ec02.pdf

klanmiko, 2017-03-09:
I've had some experience with Neural Networks, and I'd like to point out that we would not be able to run the computations fast enough on the embedded processor to train the network, we could train the network on the simulation on a computer, and give inputs based on the set of nodes we've visited, then copy the network to the bot. But that would be the only way to do it.

blhough, 2017-03-09:
Yes that would be the way.

blhough, 2017-03-15:
potentially a good starting point.
https://sourceforge.net/projects/neatvisualizers/