RoboticsClubAtUCDavis/micromouse #54: How to run SConstruct

blhough, 2017-03-10:
I get this when I try to run it. 
```
  File "SConstruct", line 14, in <module>
    sfml_libs = Split('sfml-graphics sfml-window sfml-system')
NameError: name 'Split' is not defined
```

ids1024, 2017-03-10:
Are you trying to execute it directly (`./SConstruct` or `python SConstruct`)?

The correct way to run scons is with the `scons` command.

blhough, 2017-03-10:
Okay now I get this
```
scons: Reading SConscript files ...

scons: *** An executable should have exactly one target with the suffix: .exe
File "D:\Repos\micromouse\SConstruct", line 22, in <module>
```

changing line 17 to
`OBJPREFIX='build/linux/', PROGPREFIX='build/linux/', PROGSUFFIX='.exe')`

is not working

ids1024, 2017-03-10:
Hm, I though it added the suffix on Windows. Adding a `.exe` manually to the end of the target name might fix it...

blhough, 2017-03-10:
It did not fix it either. Maybe at the next meeting you can look at it.

closed 2017-05-01 by blhough