RoboticsClubAtUCDavis/micromouse #55: Define and Implement LedRGB

blhough, 2017-03-10:
This class represents an RGB (red-green-blue) LED. An RGB LED is a light emitting diode that has three channels, one for each color. By varying the brightness of each of the three channels, an RGB LED can create any color.

Essentially an RBG LED is just 3 individual LEDs combined together. This is convenient because we already have an `Led` class.

The `LedRGB` class should have the following public members:
- [ ] Default constructor: `LedRGB();`
- [ ] Custom constructor: `LedRGB( unsigned redPin, unsigned greenPin, unsigned bluePin);`
- [ ] `void setRGB(unsigned r, unsigned g, unsigned b);`
  - Takes three values between 0-255 and sets the color accordingly.
- [ ] `void setBrightness( unsigned percent );`
  - `percent` is in the range [0,100]
- [ ] `void turnOn();`
- [ ] `void turnOff();`

Since the `Led` class has not been merged in yet, be sure to branch of your led branch.

Do not worry about inheritance yet but that will added later.

daniel9a, 2017-03-11:
i have a question so do i push the LedRGB to the led branch? or do i create a LedRGB branch and merge it to the led branch?

blhough, 2017-03-11:
(for the sake of practicing git)  we do not want to merge ledRGB into led because led is its own feature. 
With the `led` branch checked out do
`git pull origin dev` to update from dev. Then do
`git checkout -b ledrgb` to make the new branch then add your commits and push to github

PR #95 (2017-04-30): Led rgb
Closes #55 
Rebase on dev and fix minor bugs from #56 

It works great
https://goo.gl/photos/kkQnUQCpjCCcLx8QA
closed 2017-04-30 by ids1024