RoboticsClubAtUCDavis/micromouse #66: Create and Implement PIDController

blhough, 2017-03-13:
A very good intro to pid controllers: https://www.youtube.com/watch?v=4Y7zG48uHRo

wikipedia explanation: https://en.wikipedia.org/wiki/PID_controller#Fundamental_operation

We will want to be able to:
- set the constants P, I, and D.
- get the correction factor based on the current error and the delta "time"

klanmiko, 2017-03-16:
How do you plan to tune it?

blhough, 2017-03-16:
manually I guess.

I am not sure what you mean by error between encoders?

For moveForward I plan to use a controller for the speed, one for the distance, and two for keeping the mouse centered and straight. For speed and distance I use an average of the encoders so that it does not try to counter act the controllers keeping the mouse centered and straight. Eg. it's okay if one encoder say we went 99mm and the other says 101mm as long as the mouse is actual straight according to the other controllers.

klanmiko, 2017-03-16:
So we're using 2 encoders, I'm just asking how you plan to keep the robot straight and also make sure it reaches a distance accurately. By error I mean the difference in left in right encoders to keep it straight, and the difference between the average encoder value and the distance the robot should travel. 

I would suggest using 2 separate PID controllers, one for straight travel and one for distance

PR #87 (2017-04-21): Pid
closes #66 

Adds PID controller class

The reason for having the `getDelta` functor object is we might find it better to use delta distance in some instances rather than just delta time. Delta distance is the product of both time and speed so hopefully that would make the error correction more stable at varying speeds. For now only deltaTime is implemented.
closed 2017-04-21 by ids1024